ros/robot_model #102: Release ros-jade-urdf

scpeters, 2015-04-08:
I'll talk to @tfoote to see if I can help.

scpeters, 2015-04-08:
Also `urdf_parser_plugin` needs to be released into jade.

scpeters, 2015-04-08:
Waiting on https://github.com/ros-planning/geometric_shapes/issues/27

scpeters, 2015-04-22:
[geometric_shapes has been released](https://github.com/ros/rosdistro/pull/8080), so I think this is not blocked by anything anymore.

wjwwood, 2015-04-22:
I'm working on it :smile:

wjwwood, 2015-04-23:
@isucan I made two commits to the repository (other than the one that removed `resource_retriever`) and then ran a prerelease for Indigo. The prerelease ended up failing on one of the dependencies, but that looks unrelated to `robot_model`. I looked through the logs of the prerelease and I think it tested enough of it to say that it's safe to release.

I also manually reviewed the changes staged for the next indigo release and it looks ok to me.

So, I'm going to release for Indigo and then the same version into Jade. FYI

jacquelinekay, 2016-02-22:
I'm making the second release for Jade today. (Jade and Indigo are even). So I think this can be closed.

closed 2016-02-22 by jacquelinekay