ros/robot_model #103: Launch file Param Zeros error (joint_state_publisher)

fariner, 2015-04-12:
Hello,

i am trying to execute the next launch file with the "zeros" parameter of the node Joint_State_Publisher and i receive an error.

Launch File:

<launch>

	<arg name="model" />
	<arg name="gui" default="False" />
	<param name="robot_description" textfile="$(arg model)" />
	<param name="use_gui" value="$(arg gui)"/>
	<node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
        	<param name="zeros" value="{meded: 1.57075}"/>
	</node>
	<node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />
	<node name="rviz" pkg="rviz" type="rviz" args="-d $(find urdf_tutorial)/urdf.rviz" required="true" />
</launch>

URDF:

<?xml version="1.0"?>
<robot name="test">
  <link name="base_link">
    <visual>
      <geometry>
        <cylinder length="0.6" radius="0.2"/>
      </geometry>
      <material name="blue">
        <color rgba="0 0 .8 1"/>
      </material>
    </visual>
  </link>

  <link name="med">
    <visual>
      <geometry>
        <box size="0.6 .2 .1"/>
      </geometry>
      <origin rpy="0 1.57075 0" xyz="0 0 -0.3"/>
      <material name="white">
        <color rgba="1 1 1 1"/>
      </material>
    </visual>
  </link>

  <joint name="meded" type="revolute">
    <axis xyz="0 0 1"/>
    <limit effort="1000.0" lower="0.0" upper="3.1415926535" velocity="0.5"/>
    <parent link="base_link"/>
    <child link="med"/>
    <origin xyz="0 0 0.5" />
  </joint>

</robot>

Error:

ubuntu@ubuntu-HP-EliteBook-8460p:~/catkin_ws/src/visbot/urdf$ roslaunch visbot test.launch model:='$(find visbot)/urdf/my_robot.urdf' gui:=True... logging to /home/ubuntu/.ros/log/9b19dbd0-e03c-11e4-b650-101f74f8af09/roslaunch-ubuntu-HP-EliteBook-8460p-18780.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-HP-EliteBook-8460p:51617/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/zeros: meded: 1.57075
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.10
 * /use_gui: True

NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/state_publisher)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[joint_state_publisher-1]: started with pid [18798]
process[robot_state_publisher-2]: started with pid [18799]
process[rviz-3]: started with pid [18824]
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/joint_state_publisher/joint_state_publisher", line 328, in <module>
    jsp = JointStatePublisher()
  File "/opt/ros/indigo/lib/joint_state_publisher/joint_state_publisher", line 84, in __init__
    zeroval = self.zeros[name]
TypeError: string indices must be integers, not unicode
[joint_state_publisher-1] process has died [pid 18798, exit code 1, cmd /opt/ros/indigo/lib/joint_state_publisher/joint_state_publisher __name:=joint_state_publisher __log:=/home/ubuntu/.ros/log/9b19dbd0-e03c-11e4-b650-101f74f8af09/joint_state_publisher-1.log].
log file: /home/ubuntu/.ros/log/9b19dbd0-e03c-11e4-b650-101f74f8af09/joint_state_publisher-1*.log

Someone know which is the problem?

Best Regards.

jarvisschultz, 2015-10-15:
The `<param>` tag can only set "str|int|double|bool" as described [here](http://wiki.ros.org/roslaunch/XML/param). So you could change your line to
```xml
  <param name="zeros/meded" value="1.57075"/>
```
or, you could use the [`<rosparam>` tag](http://wiki.ros.org/roslaunch/XML/rosparam) to set the parameters using YAML syntax. E.g.
```xml
  <rosparam command="load">
    zeros:
      meded: 1.57075
  </rosparam>
```

That should fix your issue. The wiki really should have a description of this on the [joint_state_publisher page](http://wiki.ros.org/joint_state_publisher).

jacquelinekay, 2015-12-07:
Doesn't appear to be a bug with joint_state_publisher, closing.

closed 2015-12-07 by jacquelinekay