ros/robot_model #109: Release 1.11.8?

130s, 2015-05-30:
I'm interested in testing https://github.com/ros/robot_model/pull/108. Thanks!

130s, 2015-06-01:
[This question](http://answers.ros.org/question/209686/turtlebot-apt-get-update-has-generated-an-error-on-minimallaunch/) might also be related to the latest commit that's not yet released.

DEB is required in some tests in the downstream that's why the new release into rosdistro is necessary.

130s, 2015-06-12:
Is there any blocker for this? @isucan

130s, 2015-08-03:
ping @wjwwood

wjwwood, 2015-08-03:
This is up to @isucan. If he asks me to release it I can try to get to it sometime this week.

isucan, 2015-08-03:
I think we can release this, yes.  @wjwwood -- if you can do this, i would be very grateful.

mistoll, 2015-09-11:
Any news on this? A new release would be very helpful.

wjwwood, 2015-09-11:
Indigo: https://github.com/ros/rosdistro/pull/9433
Jade: https://github.com/ros/rosdistro/pull/9434

closed 2015-09-11 by wjwwood