ros/robot_model #110: URDF release 1.11.7 causes segfault on RVIZ

DonatasKozlovskis, 2015-06-02:
Hello!

I recently faced segfault running RVIZ  in Ubuntu arm with ROS Indigo on jetson tk1.
Isssu is caused by release 1.11.7, because downgrading URDF back to 1.11.6 solves the issue.

GDB backtrace:
<pre>
[ INFO] [1433239574.583682161]: OpenGl version: 4.4 (GLSL 4.4).
[New Thread 0xa8e363b0 (LWP 3071)]
[New Thread 0xa86363b0 (LWP 3072)]

Program received signal SIGSEGV, Segmentation fault.
is_anchored (code=code@entry=0x685460 "}", bracket_map=bracket_map@entry=0, 
    backref_map=backref_map@entry=0) at pcre_compile.c:7339
7339	pcre_compile.c: No such file or directory.

#0  is_anchored (code=code@entry=0x4e9858 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7339
#1  0xb37c80de in pcre_compile2 (pattern=<optimized out>, 
    options=<optimized out>, errorcodeptr=errorcodeptr@entry=0x0, 
    errorptr=errorptr@entry=0xbeffde44, erroroffset=0xbeffde48, 
    tables=0xb37eda70 <_pcre_default_tables> "", tables@entry=0x0)
    at pcre_compile.c:8076
#2  0xb37c8240 in pcre_compile (pattern=<optimized out>, 
    options=<optimized out>, errorptr=0xbeffde44, 
    erroroffset=<optimized out>, tables=0x0) at pcre_compile.c:7616
#3  0xac27da72 in pcrecpp::RE::Compile(pcrecpp::RE::Anchor) ()
   from /usr/lib/arm-linux-gnueabihf/libpcrecpp.so.0
#4  0xac27db7a in pcrecpp::RE::Init(std::string const&, pcrecpp::RE_Options const*) () from /usr/lib/arm-linux-gnueabihf/libpcrecpp.so.0
#5  0xac27dbc4 in pcrecpp::RE::Init(char const*, pcrecpp::RE_Options const*)
    () from /usr/lib/arm-linux-gnueabihf/libpcrecpp.so.0
#6  0xa7ab998e in pcrecpp::RE::RE(char const*) ()
   from /home/chili/catkin_ws/devel/lib/liburdf.so
#7  0xa7ab8096 in __static_initialization_and_destruction_0(int, int) ()
   from /home/chili/catkin_ws/devel/lib/liburdf.so
#8  0xa7ab8132 in _GLOBAL__sub_I_model.cpp ()
   from /home/chili/catkin_ws/devel/lib/liburdf.so
---Type <return> to continue, or q <return> to quit---
#9  0xb6fea20e in call_init (l=<optimized out>, argc=1, argv=0xbeffec44, 
    env=0xbeffec4c) at dl-init.c:78
#10 0xb6fea2a0 in _dl_init (main_map=main_map@entry=0x4f1500, argc=1, 
    argv=0xbeffec44, env=0xbeffec4c) at dl-init.c:126
#11 0xb6fed076 in dl_open_worker (a=<optimized out>) at dl-open.c:577
#12 0xb6fea140 in _dl_catch_error (objname=objname@entry=0xbeffe164, 
    errstring=errstring@entry=0xbeffe168, 
    mallocedp=mallocedp@entry=0xbeffe163, 
    operate=0xb6fece41 <dl_open_worker>, args=args@entry=0xbeffe16c)
    at dl-error.c:187
#13 0xb6feca8e in _dl_open (
    file=0x5ce864 "/opt/ros/indigo/lib/libdefault_plugin.so", 
    mode=-2147483391, 
    caller_dlopen=0xb5508e67 <Poco::SharedLibraryImpl::loadImpl(std::string const&)+46>, nsid=<optimized out>, argc=1, argv=0xbeffec44, env=0xbeffec4c)
    at dl-open.c:661
#14 0xb5094af8 in dlopen_doit (a=0xbeffe3c0) at dlopen.c:66
#15 0xb6fea140 in _dl_catch_error (objname=0x2cd74, errstring=0x2cd78, 
    mallocedp=0x2cd70, operate=0xb5094aa5 <dlopen_doit>, args=0xbeffe3c0)
    at dl-error.c:187
#16 0xb5094f48 in _dlerror_run (operate=0xb5094aa5 <dlopen_doit>, 
    args=args@entry=0xbeffe3c0) at dlerror.c:163
#17 0xb5094b82 in __dlopen (
---Type <return> to continue, or q <return> to quit---
    file=0x5ce864 "/opt/ros/indigo/lib/libdefault_plugin.so", 
    mode=<optimized out>) at dlopen.c:87
#18 0xb5508e66 in Poco::SharedLibraryImpl::loadImpl(std::string const&) ()
   from /usr/lib/libPocoFoundation.so.9
#19 0xb55091ce in Poco::SharedLibrary::SharedLibrary(std::string const&) ()
   from /usr/lib/libPocoFoundation.so.9
#20 0xb64f5556 in class_loader::class_loader_private::loadLibrary(std::string const&, class_loader::ClassLoader*) ()
   from /opt/ros/indigo/lib/libclass_loader.so
#21 0xb64f181c in class_loader::ClassLoader::loadLibrary() ()
   from /opt/ros/indigo/lib/libclass_loader.so
#22 0xb64f199a in class_loader::ClassLoader::ClassLoader(std::string const&, bool) () from /opt/ros/indigo/lib/libclass_loader.so
#23 0xb64f79ec in class_loader::MultiLibraryClassLoader::loadLibrary(std::string const&) () from /opt/ros/indigo/lib/libclass_loader.so
#24 0xb6f83162 in pluginlib::ClassLoader<rviz::ViewController>::loadLibraryForClass(std::string const&) () from /opt/ros/indigo/lib/librviz.so
</pre>

Please let me know if I can provide more information.

esteve, 2015-06-02:
We found the same bug running rviz on the Jetson, we found out that installing libpcre 1:8.35 from Ubuntu Utopic solved the segmentation fault. This may have been fixed in (http://www.pcre.org/original/changelog.txt):

Version 8.35 04-April-2014:
...
`24. Experimental support for ARM-64 and MIPS-64 has been added to the JIT compiler.`

jonbinney, 2015-09-13:
I had this same crash (well, the backtrace was pretty much the same from frame 0 up to frame 6) using robot_state_publisher on an odroid. I still had the issue when using version 1.11.7 from source, but the issue was gone when i used version 1.11.8 from source. Hopefully this issue will be gone once the arm buildfarm finishes building 1.11.8

wjwwood, 2015-09-14:
I just released `1.11.8`, which did something with the link order of `libpcre`, but I'm not sure how that impacts this. I'm just doing the releases for @isucan.

jonbinney, 2015-09-14:
Somehow 1.11.8 fixed it for me (at least from source); hopefully that wasn't just a fluke.

k-okada, 2015-10-03:
FYI: we have updated collada-dom which does not requires link order hack for `libpcre`, see 
http://lists.ros.org/lurker/message/20151003.101659.374d7c34.en.html

dchang0, 2015-12-24:
@k-okada:

Thanks for the updated collada-dom 2.4.4. Is there a .deb package for armhf? I did not see one at the links in the thread you linked to and am assuming I will need to build from source...

k-okada, 2015-12-24:
currently no, we're using ppa and currently it is not gen armhf file, if
you know how to enable this, we can try.
https://launchpad.net/~openrave/+archive/ubuntu/release

dchang0, 2015-12-24:
I don't know how (yet). Just curious: how are the i386 and amd64 .deb packages being built now? Is it fully-automated or only semi-automated?

k-okada, 2015-12-25:
We use ppa platform
https://launchpad.net/~openrave/+archive/ubuntu/release
, which is configured through this cmake setting
https://github.com/rdiankov/collada-dom/blob/master/cmake-modules/DebSourcePPA.cmake

dchang0, 2015-12-25:
Apparently you can request that Launchpad do ARM builds for you in addition to i386 and amd64:

https://dev.launchpad.net/CommunityARMBuilds

k-okada, 2015-12-25:
thanks, I sent a request.

jacquelinekay, 2016-01-28:
It doesn't seem like we need to take action in this repository, since the issue appears to be resolved by a new release of pcre. I'm closing this issue until advised otherwise.

closed 2016-01-28 by jacquelinekay

k-okada, 2016-03-10:
I am able to build arm deb, could some can copy this to ros package repository?
https://launchpad.net/~openrave/+archive/ubuntu/testing/+build/9325420
https://launchpad.net/~openrave/+archive/ubuntu/testing/+build/9325596

scpeters, 2016-03-14:
@tfoote there's an arm deb sync request here

tfoote, 2016-03-14:
Thanks @scpeters they should be there already: https://github.com/ros-infrastructure/reprepro-updater/pull/26

nuodong, 2016-07-23:
The error happens again on Kinetic  arm64 (Dragonboard 410c running  debian). Can anyone have a look? Thank!

k-okada, 2016-07-23:
Can you check version of collada-dom?

nuodong, 2016-07-26:
collada-dom is at version 2.4.0.

k-okada, 2016-07-26:
you need to use collada-dom 2.4.4, and I havent tested but it seems xenial
provide that version
http://packages.ubuntu.com/xenial/libcollada-dom2.4-dp-dev

nuodong, 2016-07-26:
@k-okada there is no package for arm64 from the url you provided above. 
 Could you give me a guide how to build and install it on DragonBoard 410c (running Debian Jessie)? Thanks.

k-okada, 2016-07-28:
how about
http://packages.ubuntu.com/xenial/arm64/libcollada-dom2.4-dp-dev/download