ros/robot_model #112: collada_urdf "double-free corruption"

pirobot, 2015-06-28:
When running the collada_urdf utility under Ubuntu 14.04 and ROS Indigo (latest Debian install), the following error message is displayed when converting even the simplest URDF file to Collada format:

     $ rosrun collada_urdf urdf_to_collada box_robot_base_only.urdf box_robot.dae

     Document successfully written to box_robot.dae
    *** Error in `/opt/ros/indigo/lib/collada_urdf/urdf_to_collada': double free or corruption (fasttop):     0x000000000191cb90 ***
    Aborted

It seems that the .dae file is created OK but the error message is rather discouraging.

skohlbr, 2015-07-05:
Same thing happens to me when I try on current .deb install on 14.04/AMD64/Indigo.

rgleichman, 2015-07-10:
I have encountered the same issue running Ubuntu 14.04 with ROS Indigo. A dae file is generated, but I don't know if it is correct. Is there any debugging info from the Ubuntu "_ has stopped unexpectedly" popup that would be useful?

wjwwood, 2015-07-10:
If someone could build it from source with `-DCMAKE_BUILD_TYPE=Debug` and then reproduce the double free while running it in `gdb` and posting the output from `bt`, that would be useful. I'm not sure if anything from the Ubuntu popup will be useful.

k-okada, 2015-07-15:
this will be resolved if you remove all pcre hack, but in order to apply #113, we have to wait for new version of collada-dom https://github.com/rdiankov/collada-dom/issues/16

jacquelinekay, 2016-01-28:
I don't see this issue with the deb anymore, probably because a new collada-dom came out, so I'm closing this issue unless anyone advises me otherwise.

closed 2016-01-28 by jacquelinekay