ros/robot_model #118: using newer collada-dom (2.4.4)

k-okada, 2015-12-10:
newer collada-dom (2.4.4) may solve collada corruption problems, https://github.com/ros/robot_model/issues/89
https://github.com/ros/robot_model/issues/112

you can download them from
https://launchpad.net/~openrave/+archive/ubuntu/release/+build/8082564, and proposed to upgrade collada-dom version in package.ros.org at 
http://lists.ros.org/lurker/message/20151016.151904.a9f138f4.ja.html

k-okada, 2015-12-10:
@tully
copy patched versoin of collada2.4 to package.ros.org
http://lists.ros.org/pipermail/ros-release/2015-October/004704.html
v amd64 (sp and dp)
https://launchpad.net/~openrave/+archive/ubuntu/release/+build/8082565 https://launchpad.net/~openrave/+archive/ubuntu/testing/+build/8338684
v i386 (sp and dp) https://launchpad.net/~openrave/+archive/ubuntu/release/+build/8082567 https://launchpad.net/~openrave/+archive/ubuntu/testing/+build/8338685
t amd64 (sp and dp) https://launchpad.net/~openrave/+archive/ubuntu/release/+build/8082564 https://launchpad.net/~openrave/+archive/ubuntu/testing/+build/8338682
t i386 (sp and dp)
 https://launchpad.net/~openrave/+archive/ubuntu/release/+build/8082566 https://launchpad.net/~openrave/+archive/ubuntu/testing/+build/8338683

k-okada, 2015-12-10:
difference between collada-dom 2.4.3 and 2.4.4 is listed at here ->
https://github.com/rdiankov/collada-dom/blob/master/changelog.rst

the problem was described in https://github.com/rdiankov/collada-dom/issues/7#issuecomment-49301822, where pcrecpp library crashes and could not get `xmlns->getURI()`, so we use urlparesr to parse url in https://github.com/rdiankov/collada-dom/commit/5939c33774ec531193a1d37a9fb52c2c2d4dec8a

tfoote, 2015-12-14:
This should be in the repositories now. And will go to public in the next sync.

k-okada, 2015-12-15:
Great !!! Thank You!!! :sushi: :sushi: :sushi: 
@garaemon, @snozawa, @orikuma, @YoheiKakiuchi

gavanderhoorn, 2015-12-21:
Is [collada-dom2.4-sp-dev causes dist-upgrade to fail on ubuntu 14.04](http://answers.ros.org/question/222892/collada-dom24-sp-dev-causes-dist-upgrade-to-fail-on-ubuntu-1404/) related to this?

k-okada, 2015-12-21:
yes, please install either dp or sp.

tfoote, 2015-12-21:
@k-okada can you ticket colladadom upstream to add a proper conflict between sp and dp? It should refuse to install both side by side early, instead of getting to the file conflict and mucking up the state of dpkg.

k-okada, 2015-12-22:
created ticket at -> https://github.com/rdiankov/collada-dom/issues/20

k-okada, 2015-12-22:
done
https://launchpad.net/~openrave/+archive/ubuntu/testing/+sourcepub/5880437/+listing-archive-extra
https://launchpad.net/~openrave/+archive/ubuntu/testing/+sourcepub/5880436/+listing-archive-extra
https://launchpad.net/~openrave/+archive/ubuntu/testing/+sourcepub/5880435/+listing-archive-extra
https://launchpad.net/~openrave/+archive/ubuntu/testing/+sourcepub/5880434/+listing-archive-extra

k-okada, 2015-12-22:
@tfoote , please copy deb in above links, current configuraiton is
collada-dom-dev provides collada-dom2.4-dp-dev
and collada-dom2.4-dp-* conflicts with collada-dom2.4-sp-*
```
k-okada@kokada-t440s:/tmp/collada/build$ sudo aptitude show collada-dom-dev
Package: collada-dom-dev                 
State: installed
Automatically installed: no
Version: 2.4.4.4-ubuntu1~trusty1
Priority: optional
Section: libs
Maintainer: OpenRAVE Testing (Testing and Releasing of OpenRAVE Packages) <openrave.testing@gmail.com>
Architecture: amd64
Uncompressed Size: 8,192 
Depends: collada-dom2.4-dp-dev
Conflicts: collada-dom-dev
Description: collada-dom 2.4.4 C++ headers and build tools
 The COLLADA Document Object Model (DOM) is an application programming interface (API) that provides a C++ object representation of a
 COLLADA XML instance document. Contact a project admin to become a developer!

k-okada@kokada-t440s:/tmp/collada/build$ sudo aptitude show collada-dom2.4-dp-dev
Package: collada-dom2.4-dp-dev           
State: installed
Automatically installed: yes
Version: 2.4.4.4-ubuntu1~trusty1
Priority: optional
Section: libs
Maintainer: OpenRAVE Testing (Testing and Releasing of OpenRAVE Packages) <openrave.testing@gmail.com>
Architecture: amd64
Uncompressed Size: 5,286 k
Depends: collada-dom2.4-dp-base
Conflicts: collada-dom2.4-sp-dev, collada-dom2.4-sp-dev, collada-dom2.4-dp-dev
Description: collada-dom 2.4.4 C++ headers and build tools
 The COLLADA Document Object Model (DOM) is an application programming interface (API) that provides a C++ object representation of a
 COLLADA XML instance document. Contact a project admin to become a developer!

k-okada@kokada-t440s:/tmp/collada/build$ sudo aptitude show collada-dom2.4-dp-base
Package: collada-dom2.4-dp-base          
State: installed
Automatically installed: yes
Version: 2.4.4.4-ubuntu1~trusty1
Priority: optional
Section: libs
Maintainer: OpenRAVE Testing (Testing and Releasing of OpenRAVE Packages) <openrave.testing@gmail.com>
Architecture: amd64
Uncompressed Size: 10.5 M
Depends: libboost-filesystem1.54.0, libboost-system1.54.0, libc6 (>= 2.14), libgcc1 (>= 1:4.1.1), libstdc++6 (>= 4.6), liburiparser1 (>=
         0.6.0), libxml2 (>= 2.7.4), zlib1g (>= 1:1.1.4)
Conflicts: collada-dom2.4-sp-base, collada-dom2.4-sp-base, collada-dom2.4-dp-base
Description: collada-dom 2.4.4 core libraries
 The COLLADA Document Object Model (DOM) is an application programming interface (API) that provides a C++ object representation of a
 COLLADA XML instance document. Contact a project admin to become a developer!

k-okada@kokada-t440s:/tmp/collada/build$ sudo aptitude show collada-dom2.4-sp-dev
Package: collada-dom2.4-sp-dev           
New: yes
State: not installed
Version: 2.4.4.4-ubuntu1~trusty1
Priority: optional
Section: libs
Maintainer: OpenRAVE Testing (Testing and Releasing of OpenRAVE Packages) <openrave.testing@gmail.com>
Architecture: amd64
Uncompressed Size: 5,286 k
Depends: collada-dom2.4-sp-base
Conflicts: collada-dom2.4-dp-dev, collada-dom2.4-dp-dev, collada-dom2.4-sp-dev
Description: collada-dom 2.4.4 C++ headers and build tools
 The COLLADA Document Object Model (DOM) is an application programming interface (API) that provides a C++ object representation of a
 COLLADA XML instance document. Contact a project admin to become a developer!

k-okada@kokada-t440s:/tmp/collada/build$ sudo aptitude show collada-dom2.4-sp-base
Package: collada-dom2.4-sp-base          
New: yes
State: not installed
Version: 2.4.4.4-ubuntu1~trusty1
Priority: optional
Section: libs
Maintainer: OpenRAVE Testing (Testing and Releasing of OpenRAVE Packages) <openrave.testing@gmail.com>
Architecture: amd64
Uncompressed Size: 10.5 M
Depends: libboost-filesystem1.54.0, libboost-system1.54.0, libc6 (>= 2.14), libgcc1 (>= 1:4.1.1), libstdc++6 (>= 4.6), liburiparser1 (>=
         0.6.0), libxml2 (>= 2.7.4), zlib1g (>= 1:1.1.4)
Conflicts: collada-dom2.4-dp-base, collada-dom2.4-dp-base, collada-dom2.4-sp-base
Description: collada-dom 2.4.4 core libraries
 The COLLADA Document Object Model (DOM) is an application programming interface (API) that provides a C++ object representation of a
 COLLADA XML instance document. Contact a project admin to become a developer!

```

gavanderhoorn, 2015-12-24:
I may be missing something but:
```
Package: collada-dom2.4-sp-base          
[..]
Conflicts: [..], collada-dom2.4-sp-base
[..]
```

Is this supposed to collide with itself?

k-okada, 2015-12-25:
I'm not sure, but it seems ok, for example liburiparser1 also has conflicts with itself.
```
$ aptitude show liburiparser1
Package: liburiparser1                   
State: installed
Automatically installed: yes
Version: 0.7.5-1ubuntu2
Priority: optional
Section: universe/libs
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Uncompressed Size: 139 k
Depends: libc6 (>= 2.14)
Conflicts: liburiparser1
Description: URI parsing library compliant with RFC 3986
 uriparser is a library that parses Uniform Resource Identifiers (URIs),
 strictly adhering to RFC 3986 where syntax for URIs is defined. URIs are used
 to identify and locate resources, like http://www.debian.org or
 ISBN:0-395-36341-1. 
 
 uriparser has the following features: 
 
 * strictly compliant to RFC 3986, implementing: + parsing + reference
   resolution + reference creation + recomposition + syntax-based normalization 
 * fast (linear input length time complexity) 
 * unicode support
Homepage: http://uriparser.sourceforge.net

```

k-okada, 2015-12-31:
new version 2.4.4.4 this has been released, and solved the problem -> http://lists.ros.org/lurker/thread/20151225.002725.4ef881c4.en.html

closed 2015-12-31 by k-okada