ros/robot_model #119: collada to urdf does not need to depend on TF

sachinchitta, 2015-12-10:
This part of the code:
https://github.com/ros/robot_model/blob/19ac39ac7324237485a27664183442a93319a269/collada_urdf/src/collada_to_urdf.cpp#L345

seems to be the only reason collada urdf depends on TF. Removing it or using urdf::Pose instead will remove the dependency.

scpeters, 2015-12-11:
The `tf::Quaternion` could be replaced easily, but the `tf::Matrix3x3` would need Eigen or some other dependency.

jacquelinekay, 2016-01-28:
Options:

1. Introduce dependency on Eigen.
2. Replace datatypes with `urdf::Pose`.
  - [urdf::Rotation](https://github.com/ros/urdfdom_headers/blob/master/urdf_model/include/urdf_model/pose.h#L100) represents quaternions and accepts quaternion (x, y, z, w) coordinates in the constructor.
  - [Line 349](https://github.com/ros/robot_model/blob/19ac39ac7324237485a27664183442a93319a269/collada_urdf/src/collada_to_urdf.cpp#L349) could be replaced with `urdf::Rotation::GetInverse()`, since the transpose of of a rotation matrix is also its inverse.
  - Some extra code will be needed to then convert from quaternion or RPY representation to a 3x3 rotation matrix ([argh](http://www.euclideanspace.com/maths/geometry/rotations/conversions/quaternionToMatrix/)).
3. Get rid of the code that needs mathematical representation of rotations, since it is necessary only for debugging purposes.
4. Do nothing.

I'd rather not take option 3, since I can see how the debug output would be useful.

2 is appealing because much of the necessary code is already there. However, the last step introduces some evil code duplication of slightly tricky mathematical formulas into either `robot_model` or `urdfdom_headers`.

Comparing 1 and 4: I can understand why a dependency on tf is unappealing, because there of the extra code in tf for transport. On one hand, Eigen is outside of the ROS ecosystem, but on the other hand, it's stable and well-supported. And, for what it's worth, [collada_urdf](https://github.com/ros/robot_model/blob/indigo-devel/collada_urdf/package.xml#L44) already has an indirect dependency on Eigen via [`geometric_shapes`](https://github.com/ros-planning/geometric_shapes/blob/indigo-devel/package.xml).

So I'm leaning towards @scpeters's suggestion of replacing tf with eigen.

jacquelinekay, 2016-03-25:
I'm going to prototype replacing the dependency on tf with Eigen.

sloretz, 2017-05-11:
Thanks for the issue! `collada_parser` has been moved to a new repository. I've copied the issue to ros/collada_urdf#6

closed 2017-05-11 by sloretz