ros/robot_model #126: Kinetic Release

wjwwood, 2016-04-04:
Can this be released into Kinetic (even just initially)? It's blocking a first release of `rviz`.

jacquelinekay, 2016-04-04:
This repo is blocked by geometric_shapes and random_numbers. They are pretty stable, but just in case I will ping @davetcoleman, @mikeferguson etc--is it okay if we release those packages into Kinetic?

jacquelinekay, 2016-04-04:
asking for forgiveness instead of permission here:
https://github.com/ros/rosdistro/pull/10992
https://github.com/ros/rosdistro/pull/10993

@wjwwood I'll release robot_model now. Cheers.

jacquelinekay, 2016-04-04:
initial release is done: https://github.com/ros/rosdistro/pull/10994

I'll release again if/when I finish those features I was trying to get into Kinetic. :)

closed 2016-04-04 by wjwwood