ros/robot_model #130: when I install ROS on RaspberryPi  3 I got below problem

Tom-Shen, 2016-04-24:
how do I do?

========================================================================
==> Processing catkin package: 'urdf'
==> Creating build directory: 'build_isolated/urdf'
==> Building with env: '/opt/ros/indigo/env.sh'
==> cmake /home/pi/ros_catkin_ws/src/robot_model/urdf -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/urdf -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -G Unix Makefiles in '/home/pi/ros_catkin_ws/build_isolated/urdf'
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   thread
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findurdfdom.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "urdfdom", but
  CMake did not find one.

  Could not find a package configuration file provided by "urdfdom" with any
  of the following names:

    urdfdomConfig.cmake
    urdfdom-config.cmake

  Add the installation prefix of "urdfdom" to CMAKE_PREFIX_PATH or set
  "urdfdom_DIR" to a directory containing one of the above files.  If
  "urdfdom" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/ros_catkin_ws/build_isolated/urdf/CMakeFiles/CMakeOutput.log".
<== Failed to process package 'urdf': 
  Command '['/opt/ros/indigo/env.sh', 'cmake', '/home/pi/ros_catkin_ws/src/robot_model/urdf', '-DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/urdf', '-DCMAKE_INSTALL_PREFIX=/opt/ros/indigo', '-DCMAKE_BUILD_TYPE=Release', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/pi/ros_catkin_ws/build_isolated/urdf && /opt/ros/indigo/env.sh cmake /home/pi/ros_catkin_ws/src/robot_model/urdf -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/urdf -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'

Command failed, exiting.

========================================================================

Tom-Shen, 2016-04-24:
I fixed it

closed 2016-04-24 by Tom-Shen

AloshkaD, 2016-06-30:
Hi Tom, I'm gitting the same error, how did you fix it?

wallEVA96, 2017-04-15:
could u  tell me how to solve this issue

vasuvishnu, 2017-11-24:
Hi Tom,
I'm getting the same error, can u tell me how did u fixed that errror.
i'm trying for two days, please help me.