ros/robot_model #132: collada_urdf: Box geometries appear to have wrong size

skohlbr, 2016-04-26:
As also previously noted on the [OpenRAVE mailing list](http://openrave-users-list.185357.n3.nabble.com/Box-geometry-in-URDF-does-not-scale-in-DAE-td4027346.html), it appears that box geomtries are parsed not correctly, resulting in box geometries having half the size they should. See screenshots below for illustration:

URDF in rviz:
[image]

Generated DAE in Blender:
[image]

Note that all box geometries are scaled wrongly, while all mesh geoms look correct.

This is using Ubuntu 14.04/AMD64/Indigo/latest collada_urdf from .debs.

jacquelinekay, 2016-04-26:
thanks for reporting this; I opened #133 as a potential fix.

clalancette, 2017-02-01:
I just pushed Jackie's fix into the upstream repository, which should fix this problem.  If you are still having problems, please feel free to re-open this issue.

closed 2017-02-01 by clalancette