ros/robot_model #136: joint_state_publisher: When using gui (use_gui=true) the joint_state_publisher node does not exit

vincentrou, 2016-05-26:


vincentrou, 2016-05-26:
Ubuntu 16.04 and ROS kinetic

tlind, 2016-08-25:
@jacquelinekay In PR #138, you write:

> My current theory is that Qt is intercepting the CTRL-C and not passing it on to rospy.

That is correct. The following discussion outlines a few ways on how this can be handled:
http://stackoverflow.com/questions/4938723/what-is-the-correct-way-to-make-my-pyqt-application-quit-when-killed-from-the-co

In my own pyQt applications, I am calling

        signal.signal(signal.SIGINT, signal.SIG_DFL)
        signal.signal(signal.SIGTERM, signal.SIG_DFL)
before initializing the QtApplication object. Note that this will cause the GUI to exit non-gracefully upon CTRL+C, i.e. without calling any finalizers etc.

vincentrou, 2017-07-28:
This was fix by PR #139

closed 2017-07-28 by vincentrou