ros/robot_model #141: can't compile at gcc 6

Yayi, 2016-07-21:


AlexDaniel, 2016-09-26:
Yeah.

NigoroJr, 2016-11-01:
I'm not sure what error messages you get, but I couldn't compile with gcc 5.4.0 either on my Gentoo box because of errors basically saying it can't find C++11 classes.

It turns out that urdfdom_headers 1.0.0 converted from boost to C++11 and since my system had urdfdom_headers 1.0.0 because of requirements for Gazebo, collada_parser couldn't compile the included headers. I worked around this by cloning urdfdom_headers and checking out to 0.4.2, installing its header files to a temporary location and having collada_parser find them by using the `HINTS` directive in `CMakeLists.txt`. I'm a ROS newbie so that was the easiest workaround that I could think of but I think there are proper ways of specifying the dependency packages' versions.

wjwwood, 2016-11-01:
I need more information (like a compiler error) to help you guys. I'll reopen with more information.

closed 2016-11-01 by wjwwood

AlexDaniel, 2016-11-01:
@wjwwood why not just get the latest gcc and try to compile it yourself?

wjwwood, 2016-11-01:
@AlexDaniel because @Yayi already did that and I'm super busy. I don't know why you'd report a problem and not include the details...

NigoroJr, 2016-11-04:
Just as follow-up, I switched to using portage rather than `catkin_make_isolated` and the issue completely disappeared. It was probably me doing something strange.