ros/robot_model #146: [joint_state_publisher] Change VBoxLayout to Gridlayout?

bmagyar, 2016-08-24:
Hey guys,
I've use this node so many times to quickly test stuff but sometimes the robots I'm working with have simply too many joints and I can't keep buying bigger screen! ( :D )

I changed the GUI yesterday slightly to use a max items per column. For now I've hardcoded it to 17 because that seemed to work for me. 

Any better ideas how to do this dynamically? I looked around but no quick search revealed a simple solution. Perhaps I could make my magic number a parameter and have a default disabled behaviour for it?

Any interest in getting something for this merged? I could provide a PR.
[image]

wjwwood, 2016-08-24:
Sounds like a reasonable idea. I looked a bit too for a "columnizer" (jquery uses this word, but maybe it's the wrong thing in the qt world), but I couldn't find anything either. It would be nice to have it be dynamic, but a parameter in the gui or on the cli would be ok too probably.

bmagyar, 2016-08-24:
The on-gui option didn't even come to my mind, good idea!

I think the best would be to have the GUI option as well as the cli one.
Cli could be used to initialise the gui part which would control the
layout. The default value of course could replicate the current behaviour.

I can implement this tomorrow morning ;)

tlind, 2016-08-25:
I also noticed recently that the joint state publisher GUI does not scroll vertically anymore following the switch to Qt5. This is a major showstopper if there are lots of joints (in my case, I have like 40 or so, which don't fit on a single screen).

I was just about to create a PR to wrap the main widget into a `ScrollArea` to reenable that lost functionality, when I came across this issue. The screenshot posted by @bmagyar certainly looks nice, if the number of columns could also be adjustable?

bmagyar, 2016-08-26:
Yes, I'm on it ;) spent quite some time with it yesterday but haven't fully
cracked it yet! I'll submit a PR today for discussion

bmagyar, 2016-08-31:
Hiya,
Please come over to the PR and hit me with your comments! :)

VictorLamoine, 2016-12-22:
@wjwwood or @bmagyar you can close this issue

closed 2016-12-22 by wjwwood