ros/robot_model #149: collada_urdf 1.12.4 fails to build on Wily and Jessie

tfoote, 2016-08-26:
It looks like an eigen dependency issue: 

http://build.ros.org/view/Kbin_uW64/job/Kbin_uW64__collada_urdf__ubuntu_wily_amd64__binary/22/console
```
1.12.4/src/collada_urdf.cpp
In file included from /opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_containers.h:40:0,
                 from /opt/ros/kinetic/include/geometric_shapes/mesh_operations.h:41,
                 from /tmp/binarydeb/ros-kinetic-collada-urdf-1.12.4/src/collada_urdf.cpp:89:
/opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_vector_container.h:40:22: fatal error: Eigen/Core: No such file or directory
compilation terminated.
CMakeFiles/collada_urdf.dir/build.make:57: recipe for target 
```

http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__collada_urdf__debian_jessie_amd64__binary/25/console
```
[ 66%] Building CXX object CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o
/usr/lib/ccache/c++   -DASSIMP_EXPORT_API -DASSIMP_UNIFIED_HEADER_NAMES -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"collada_urdf\" -Dcollada_urdf_EXPORTS -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2  -fPIC -I/tmp/binarydeb/ros-kinetic-collada-urdf-1.12.4/include -I/usr/include/assimp -I/usr/include/collada-dom2.4/1.5 -I/usr/include/collada-dom2.4 -I/opt/ros/kinetic/include    -std=c++11 -o CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o -c /tmp/binarydeb/ros-kinetic-collada-urdf-1.12.4/src/collada_urdf.cpp
In file included from /opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_containers.h:40:0,
                 from /opt/ros/kinetic/include/geometric_shapes/mesh_operations.h:41,
                 from /tmp/binarydeb/ros-kinetic-collada-urdf-1.12.4/src/collada_urdf.cpp:89:
/opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_vector_container.h:40:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
CMakeFiles/collada_urdf.dir/build.make:57: recipe for target 
```

tfoote, 2016-08-26:
#143 looks releated

dirk-thomas, 2016-09-02:
This needs to be resolved in order to sync Kinetic.

wjwwood, 2016-09-02:
I think that this package was previously depending on `geometric_shapes` to export the Eigen flags, but I think this recently released change broke the exporting at least on Wily and Jessie: https://github.com/ros-planning/geometric_shapes/pull/46

Not sure what the right fix is.

wjwwood, 2016-09-02:
Actually that's been out since July 31st. Not sure anymore.

wjwwood, 2016-09-02:
Looking at the traceback of the issue, I think `eigen_stl_containers` should export eigen as a transitive dependency and `geometric_shapes` should export `eigen_stl_containers` as a transitive dependency. It looks like the first hop is there: https://github.com/ros/eigen_stl_containers/blob/master/CMakeLists.txt

And the link from above implies it is also there in `geometric_shapes`, but one and/or both exports of Eigen may just be broken.

I still don't see why it just started failing.

wjwwood, 2016-09-02:
I suppose it's possible that both `geometric_shapes` and `eigen_stl_containers` export of eigen has been broken for a while, but yet another package was masking that by properly exporting eigen until recently.

Either way, it doesn't appear to me that `collada_urdf` uses Eigen directly, at least it doesn't include the headers directly, so this should be fixed in another package.

k-okada, 2016-09-26:
I look into this trouble and may found solution https://github.com/ros/eigen_stl_containers/pull/6
May be wily uses eigen 3.2 makes this problem. xenal uses 3.5 and provide correct EIGEN3_INCLUDE_DIRS.

sloretz, 2017-02-14:
@wjwwood Can this be closed now that ros/eigen_stl_containers#6 has been merged?

clalancette, 2017-02-14:
It looks to me like the issue on the build farm was resolved quite a while ago (the wily package has been successfully built since September).  Also, I was just able to build this locally on Wily.  So I'm going to close out this issue for now; if this reoccurs, feel free to re-open.

closed 2017-02-14 by clalancette