ros/robot_model #152: *SharedPtr macros?

bmagyar, 2016-09-02:
Hiya,

This week we had an issue where the guy was compiling everything from latest (unreleased) sources and found that `ros_control` had `boost::shared_ptr<urdf::XY>` instead of `urdf::XYConstSharedPtr` __which eases migration from `boost` stuff to `std`__.

[Here](https://github.com/ros-controls/ros_controllers/pull/235/files/72462aeb68bd7e0600ad2d8c09d81b94793e8877#r77217517)'s the related issue and I discovered that urdf::Model didn't have this macro. All the stuff in `urdfdom_headers` have it, do you think it'd make sense to add the macro here too?

wjwwood, 2016-09-02:
It wouldn't be a macro, but a typedef or a type alias. I'd review a pr to add this.

de-vri-es, 2016-10-02:
This can be closed now that #153 is merged I think?

bmagyar, 2016-10-02:
Indeed

closed 2016-10-02 by bmagyar