ros/robot_model #166: collada_parser build failure on debian jessie

jabberwox, 2016-11-10:
Collada_parser failed building on my raspberry pi 3 (debian jessie). I ran into the following error:

```
....
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2434:59: error: request for member ‘end’ in ‘((urdf::ColladaModelReader*)this)->urdf::ColladaModelReader::_model.boost::shared_ptr<T>::operator-><urdf::ModelInterface>()->urdf::ModelInterface::joints_’, which is of non-class type ‘int’
         if (_model->joints_.find(name) == _model->joints_.end()) {
                                                           ^
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2437:38: error: request for member ‘find’ in ‘((urdf::ColladaModelReader*)this)->urdf::ColladaModelReader::_model.boost::shared_ptr<T>::operator-><urdf::ModelInterface>()->urdf::ModelInterface::joints_’, which is of non-class type ‘int’
             pjoint = _model->joints_.find(name)->second;
                                      ^
CMakeFiles/collada_parser.dir/build.make:54: recipe for target 'CMakeFiles/collada_parser.dir/src/collada_parser.cpp.o' failed
make[2]: *** [CMakeFiles/collada_parser.dir/src/collada_parser.cpp.o] Error 1
CMakeFiles/Makefile2:91: recipe for target 'CMakeFiles/collada_parser.dir/all' failed
make[1]: *** [CMakeFiles/collada_parser.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'collada_parser': 
  Command '['/opt/ros/indigo/env.sh', 'make', '-j1']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/pi/ros_catkin_ws/build_isolated/collada_parser && /opt/ros/indigo/env.sh make -j1
```

Re-running the make command gives:

```
[ 50%] Building CXX object CMakeFiles/collada_parser.dir/src/collada_parser.cpp.o
CMakeFiles/collada_parser.dir/build.make:54: recipe for target 'CMakeFiles/collada_parser.dir/src/collada_parser.cpp.o' failed
CMakeFiles/Makefile2:91: recipe for target 'CMakeFiles/collada_parser.dir/all' failed
Makefile:127: recipe for target 'all' failed
```

I raised the question the issue also [here](http://answers.ros.org/question/247523/collada_parser-build-failed-on-raspberry-pi-3-jessie-debian-ros-indigo/).

wjwwood, 2016-11-10:
Can you post the ROS version you're using and/or the version of `robot_model`? Looks like the Kinetic Debian Jessie job for armv8 is passing:

http://build.ros.org/view/Kbin_djv8_dJv8/job/Kbin_djv8_dJv8__collada_parser__debian_jessie_arm64__binary/

The architecture might be slightly different from what the rpi3 has, but it should have similar dependencies.

nickxiang0306, 2016-11-24:
I got the same problem. I am using rosdisto: indigo, rosversion: 1.11.20. And the version of robot_model is indigo-devel.

JeremyBYU, 2017-02-12:
Same issue.  Following WIKI on building ROS on indigo. I'm not sure the issue is specific to robot_model.  

```
In file included from /usr/local/include/urdf_model/joint.h:43:0,
                 from /usr/local/include/urdf_model/link.h:44,
                 from /usr/local/include/urdf_model/model.h:42,
                 from /home/pi/ros_catkin_ws/src/robot_model/collada_parser/include/collada_parser/collada_parser.h:44,
                 from /home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:58:
/usr/local/include/urdf_model/pose.h: In member function ‘void urdf::Vector3::init(const string&)’:
/usr/local/include/urdf_model/pose.h:78:25: error: ‘stod’ is not a member of ‘std’
           xyz.push_back(std::stod(pieces[i]));

```
I googled that error and its seems that std:stod is only available for [c++11](http://www.cplusplus.com/forum/general/98579/).

sloretz, 2017-02-14:
@JeremyBYU the c++11 issue should have been fixed by #181. Would you mind trying again with the latest code on kinetic-devel?

sloretz, 2017-05-11:
Thanks for the issue! `collada_parser` has been moved to a new repository. I've copied the issue to ros/collada_urdf#7

closed 2017-05-11 by sloretz