ros/robot_model #169: URDF to Collada: joint coordinate system not observed?

sirop, 2016-12-06:
I have a problem similar to this http://answers.ros.org/question/221675/unable-to-move-links-within-urdf-file/ , it is only that I convert from URDF to Collada.

Original URDF as seen in RViz:
[image]

After converting to Collada as seen in Openrave:
[image]

If I judge solely by eyesight, it seems to me that in the converted Collada file the visual elements are rendered  not w.r.t. (local) joint coordinate systems.

I am using Indigo:
```
$ dpkg -l | grep indigo-collada
ii  ros-indigo-collada-parser                             1.11.11-0trusty-20160627-175110-0700                amd64        This package contains a C++ parser for the Collada robot description format.
ii  ros-indigo-collada-urdf                               1.11.11-0trusty-20161111-194651-0800                amd64        This package contains a tool to convert Unified Robot Description Format (URDF) documents into COLLAborative Design Activity (COLLADA) documents.
```
Is this a known problem, maybe, already solved in Kinectic?

Thanks.

wjwwood, 2017-01-26:
I'm not aware of this issue being reported before, I do not have time to reproduce it on Kinetic, if you find time can you please comment here?

sirop, 2017-02-08:
Similar case:
http://openrave-users-list.185357.n3.nabble.com/urdf-to-collada-changing-fixed-joint-location-tt4027749.html

sloretz, 2017-05-11:
Thanks for the issue! `collada_urdf` has been moved to a new repository. I've copied the issue to ros/collada_urdf#8

closed 2017-05-11 by sloretz