ros/robot_model #171: URDF compatibility build error on debian jessie

mryellow, 2016-12-10:
Believe this might be an edge-case in `urdf_compatibility.h` which belongs here.

```
src/ros_controllers/diff_drive_controller/src/diff_drive_controller.cpp:61:36: error: ‘LinkConstSharedPtr’ in namespace ‘urdf’ does not name a type
 static bool isCylinder(const urdf::LinkConstSharedPtr& link)
```

`liburdfdom-dev` with Jessie is `0.3.0-1+b1`, which I guess means only Model

wjwwood, 2016-12-11:
What version of ROS are you using?

mryellow, 2016-12-11:
Kinetic, source install built from a fresh desktop rosinstall generator.

Looks like all the involved packages are at their most recent state, with no new changes.

```
- tar:
    local-name: ros/ros
    uri: https://github.com/ros-gbp/ros-release/archive/release/kinetic/ros/1.13.4-0.tar.gz
    version: ros-release-release-kinetic-ros-1.13.4-0
```

```
- tar:
    local-name: robot_model/robot_model
    uri: https://github.com/ros-gbp/robot_model-release/archive/release/kinetic/robot_model/1.12.5-0.tar.gz
    version: robot_model-release-release-kinetic-robot_model-1.12.5-0
- tar:
    local-name: robot_model/urdf
    uri: https://github.com/ros-gbp/robot_model-release/archive/release/kinetic/urdf/1.12.5-0.tar.gz
    version: robot_model-release-release-kinetic-urdf-1.12.5-0
```

```
- git:
    local-name: ros_control
    uri: https://github.com/ros-controls/ros_control.git
    version: kinetic-devel
- git:
    local-name: ros_controllers
    uri: https://github.com/ros-controls/ros_controllers.git
    version: kinetic-devel
- git:
    local-name: realtime_tools
    uri: https://github.com/ros-controls/realtime_tools.git
    version: kinetic-devel
- git:
    local-name: control_toolbox
    uri: https://github.com/ros-controls/control_toolbox.git
    version: kinetic-devel
```

wjwwood, 2017-01-04:
Sorry for the delay in responding.

What version of `liburdfdom-dev` (see https://github.com/ros/rosdistro/blob/90ef1ed07cf065b2e6d819abdcdeff4881d6e7ef/rosdep/base.yaml#L2773) do you have installed?

It builds on our farm: http://repositories.ros.org/status_page/ros_kinetic_dj.html?q=diff_drive_controller

So I guess it must be a configuration of your machine?

mryellow, 2017-01-04:
`liburdfdom-dev` with Jessie is `0.3.0-1+b1`.

> So I guess it must be a configuration of your machine?

With it being a source install, entirely possible. Everything does appear to be up-to-date with deployed versions, but you never know might be something upstream which confuses the whole situation, a few packages interacting here.

Haven't had a look at this for a number of weeks, if it's not coming up in your builds then might be safe to ignore and I'll find the time to figure out what is special about my build.

wjwwood, 2017-01-26:
I'm going to close this since I could not reproduce and @mryellow hasn't had time to follow up. Please comment here if this needs to be reopened.

closed 2017-01-26 by wjwwood

mryellow, 2017-01-27:
Sorry haven't had a chance to touch any of this stuff for awhile.

Cleared root workspace and re-run rosinstall generator `desktop_full` which updated `robot_model` from `1.12.5` to `1.12.6`. Along the way patched dependency issues in `image_pipeline`.

Added extra packages outside of `desktop_full` for `diff_drive_controller`.

```
- git:
    local-name: ros_control
    uri: https://github.com/ros-controls/ros_control.git
    version: kinetic-devel
- git:
    local-name: ros_controllers
    uri: https://github.com/ros-controls/ros_controllers.git
    version: kinetic-devel
- git:
    local-name: realtime_tools
    uri: https://github.com/ros-controls/realtime_tools.git
    version: kinetic-devel
- git:
    local-name: control_toolbox
    uri: https://github.com/ros-controls/control_toolbox.git
    version: kinetic-devel
```

Result:

```
src/ros_controllers/diff_drive_controller/src/diff_drive_controller.cpp:61:36: error: ‘LinkConstSharedPtr’ in namespace ‘urdf’ does not name a type
 static bool isCylinder(const urdf::LinkConstSharedPtr& link)
```

No idea... Think I'll try it on actual hardware with binary packages rather than dev machine.

wjwwood, 2017-01-27:
Ok, let me know if you have a chance to narrow it down. One other thing to point out is that the build farm uses the released version of the dependencies and it seems like you might be using the default branch from some of the dependencies. That could be an important difference, idk.

mryellow, 2017-01-27:
> That could be an important difference, idk.

Yeah I try the released github as source when things get strange, though these `kinetic-devel` branches seem to match the releases.

> narrow it down

Found something!

https://github.com/ros-controls/ros_controllers/pull/246

Adding `#include <urdf/urdfdom_compatibility.h>` is needed on the `diff_drive_controller` end.

Think I assumed that `#include <urdf_parser/urdf_parser.h>` would include that and no changes would be needed on dependent packages...

wjwwood, 2017-01-27:
Hmm, ok that package is also not failing on the farm but it sounds like you're on the right track. Let me know if you'd like this reopened.

I don't think it requires changes, but you do need that header if you make use of the new symbols (rather than the `boost::*` symbols. Maybe it's a transitive problem, i.e. some other package's header uses the symbols but doesn't include the header, and so including the header in the diff drive controller fixes it. I don't know, but I don't have time to track it down atm.

mryellow, 2017-01-27:
> Let me know if you'd like this reopened.

Can leave it with `ros_controllers`, depending on this bit I don't really understand completely. My C++ OO understanding is somewhat hacky... 

When they include `urdf_parser/urdf_parser.h` shouldn't that be including `urdf/urdfdom_compatibility.h` without the need for any changes?

> Maybe it's a transitive problem

Seems plausible, I notice `1.12.5 - 1.12.6` is when you switch from `model.h` to `urdfdom_compatibility.h` and I started looking at this with `1.12.5`.

mryellow, 2017-01-27:
This comment breaks down where the header is and isn't included:

https://github.com/ros-controls/ros_controllers/pull/246#issuecomment-275628491

wjwwood, 2017-01-27:
The header is provided by `urdf`, and `urdf` depends on `urdfdom` and `urdfdom_headers` and so it makes sense they cannot include it.