ros/robot_model #175: urdf_to_collada crashes with negative cylinder lengths

mrjogo, 2017-01-09:
When urdf_to_collada is run with the following link, it crashes because of the negative length:

```xml
<link name="link0">
  <visual>
    <geometry>
      <cylinder length="-0.005" radius="0.01"/>
    </geometry>
  </visual>
</link>
```

The error message is
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

It's possible that negative lengths are not valid URDF, but it is not specified in the [URDF documentation](http://wiki.ros.org/urdf/XML/link) and the other systems that use URDF happily interpret the length as starting from the origin and going in the negative direction.

clalancette, 2017-01-10:
I took a quick look at this.  It turns out that the problem is down in geometric_shapes, where we have this code:

  double h = cylinder.length;
  ...
  unsigned int h_num = ceil(h / circle_edge);
  ...
  for (unsigned int i = 0; i < h_num - 1 ; ++i)
    for(unsigned int j = 0; j < tot; ++j)
      vertices.push_back(Eigen::Vector3d(r * cos(phi + phid * j), r * sin(phi + phid * j), h / 2 - (i + 1) * hd));

So we are taking a negative double, dividing it by some (positive) amount, and then assigning the result to an unsigned.  This turns out to be a *huge* number (instead of the small negative number), which we then end up iterating over and doing a vector push_back, using up a crazy amount of memory.  A simple fix here is call fabs() on h before calculating h_num, which will avoid this particular problem, but may or may not actually "work".  I'll get in touch with the geometric_shapes maintainers and see if they have any advice.

wjwwood, 2017-01-26:
Close this in favor of https://github.com/ros-planning/geometric_shapes/issues/64

closed 2017-01-26 by wjwwood