ros/robot_model #176: Grid layout with 0 free joints causes crash

VictorLamoine, 2017-01-11:
If you load a robot model with 0 free joints the joint state publisher crashes:

```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher", line 413, in <module>
    jsp = JointStatePublisher()
  File "/opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher", line 127, in __init__
    self.gui = JointStatePublisherGui("Joint State Publisher", self, num_rows)
  File "/opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher", line 304, in __init__
    self.positions = self.generate_grid_positions(len(sliders), self.num_rows)
  File "/opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher", line 383, in generate_grid_positions
    positions = [(y, x) for x in range(int((math.ceil(float(num_items) / num_rows)))) for y in range(num_rows)]
ZeroDivisionError: float division by zero
```

#150

VictorLamoine, 2017-01-11:
@bmagyar would you be so kind as to open a PR and fix this?

bmagyar, 2017-01-19:
Sure, I am looking at it now. Is there a reason you have it open or just found this corner-case by accident?

VictorLamoine, 2017-01-19:
By accident!

PR #178 (2017-01-19): Fix crash on 0 free joints, opens empty window
Fixes #176 by simply skipping the math when having to return 0 rows. GUI opens up empty as it did before the gridlayout changes.
closed 2017-01-19 by wjwwood