ros/robot_model #188: floating joints fail

kopp, 2017-03-05:
I would like to use a `floating` joint as described [here](http://wiki.ros.org/urdf/XML/joint).

I used the following urdf file

```
<?xml version="1.0"?>
<robot name="two_frames">
    <link name="base_link" />
    <link name="dependent_link" />
    <joint name="dependent_to_base_joint" type="floating">
        <parent link="base_link" />
        <child link="dependent_link" />
        <origin xyz="0 0 1" rpy="0 0 0" /> <!-- set some initial value -->
    </joint>
</robot>
```

and then launch that with

```
<launch>
  <param name="robot_description" textfile="$(find my_package)/urdf/two_frames.urdf" />
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" >
    <param name="use_gui" value="true" />
  </node>
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen"/>
</launch>
```

I get the output

    [ WARN] [1488741790.664110960]: Converting unknown joint type of joint 'dependent_to_base_joint' into a fixed joint
    [ INFO] [1488741790.676139057]: Floating joint. Not adding segment from base_link to dependent_link. This TF can not be published based on joint_states info

Is [this answer](http://answers.ros.org/question/74626/i-get-a-warnning-when-trying-to-use-a-floating-joint/) true and `floating` is simply not longer supported?  If so, please remove it from the docs (or add it as it would be quite useful...).

sloretz, 2017-04-06:
... doing a little investigating.

[ros/urdfdom](https://github.com/ros/urdfdom) is aware of floating joints [joing.cpp:519](https://github.com/ros/urdfdom/blob/8b27e0b6fbddbb4cc670156961bd52b4d5ea632b/urdf_parser/src/joint.cpp#L519), but kdl_parser is not [kdl_parser.cpp:89](https://github.com/ros/robot_model/blob/409c4b923c38582ba595c7fc2a0536f0ee60045c/kdl_parser/src/kdl_parser.cpp#L89). Planar joints are also missing from kdl_parser. I think the root issue is KDL only supports joints with 1 degree of freedom. [Documentation here](http://docs.ros.org/api/orocos_kdl/html/classKDL_1_1Joint.html#ad05f78142c8b73c3977f038a7a2c544e)

The second error in your output is from [robot_state_publisher.cpp:66](https://github.com/ros/robot_state_publisher/blob/b9da64fd509521c9f24f9be634054d8366174180/src/robot_state_publisher.cpp#L66). It was introduced in [this commit](https://github.com/ros/robot_state_publisher/commit/8235bce9b65ee9723a4f0a0e03d7258737cf07ff). The error message seems to say won't publish the transform because [sensor_messages/JointState](http://docs.ros.org/api/sensor_msgs/html/msg/JointState.html) doesn't have enough information to publish the transform. robot_state_publisher expects to publish updated joint transforms based on input on it's topic `joint_states`.

Related to issue #86 

I think your best option is to use robot_state_publisher to publish transforms from JointState messages for all the other joints in your model, and then create your own node which publishes the tf transform from `base_link` to `dependent_link` using whatever sensors on the bot provide that information.

kopp, 2017-04-07:
So my best option is to basically not use the `joint_state_publisher` for these joints?

sloretz, 2017-04-07:
@kopp Yes. I think your best option is to publish tf transforms for joints that have more than 1 degree of freedom (planar, floating) yourself.

The purpose of the joint_state_publisher is to publish `sensor_msgs/JointState` messages, but that message type cannot describe more than 1 degree of freedom.

I'll close this issue because I don't think it will be fixed. Feel free to reopen.

closed 2017-04-07 by sloretz