ros/robot_model #189: ROS indigo build failed on debian 8

ita93, 2017-03-13:
Currently, I tried to install ROS indigo version on debian 8 (jessie), I also apply patch to fix c++ 11 issues, but I procedure another issue late to URDF. The log as before:
`[ 33%] Building CXX object CMakeFiles/urdf.dir/src/model.cpp.o
In file included from /home/ros/ros_catkin_ws/src/robot_model/urdf/include/urdf/model.h:43:0,
                 from /home/ros/ros_catkin_ws/src/robot_model/urdf/src/model.cpp:37:
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:78:25: error: ‘ModelInterface’ was not declared in this scope
 typedef std::shared_ptr<ModelInterface> ModelInterfaceSharedPtr;
                         ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:78:25: note: suggested alternative:
In file included from /home/ros/ros_catkin_ws/src/robot_model/urdf/include/urdf/model.h:42:0,
                 from /home/ros/ros_catkin_ws/src/robot_model/urdf/src/model.cpp:37:
/usr/local/include/urdf_model/model.h:48:7: note:   ‘urdf::ModelInterface’
 class ModelInterface
       ^
In file included from /home/ros/ros_catkin_ws/src/robot_model/urdf/include/urdf/model.h:43:0,
                 from /home/ros/ros_catkin_ws/src/robot_model/urdf/src/model.cpp:37:
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:78:39: error: template argument 1 is invalid
 typedef std::shared_ptr<ModelInterface> ModelInterfaceSharedPtr;
                                       ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:78:64: error: invalid type in declaration before ‘;’ token
 typedef std::shared_ptr<ModelInterface> ModelInterfaceSharedPtr;
                                                                ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:79:31: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
 typedef std::shared_ptr<const ModelInterface> ModelInterfaceConstSharedPtr;
                               ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:79:45: error: template argument 1 is invalid
 typedef std::shared_ptr<const ModelInterface> ModelInterfaceConstSharedPtr;
                                             ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:79:75: error: invalid type in declaration before ‘;’ token
 typedef std::shared_ptr<const ModelInterface> ModelInterfaceConstSharedPtr;
                                                                           ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:80:23: error: ‘ModelInterface’ was not declared in this scope
 typedef std::weak_ptr<ModelInterface> ModelInterfaceWeakPtr;
                       ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:80:23: note: suggested alternative:
In file included from /home/ros/ros_catkin_ws/src/robot_model/urdf/include/urdf/model.h:42:0,
                 from /home/ros/ros_catkin_ws/src/robot_model/urdf/src/model.cpp:37:
/usr/local/include/urdf_model/model.h:48:7: note:   ‘urdf::ModelInterface’
 class ModelInterface
       ^
In file included from /home/ros/ros_catkin_ws/src/robot_model/urdf/include/urdf/model.h:43:0,
                 from /home/ros/ros_catkin_ws/src/robot_model/urdf/src/model.cpp:37:
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:80:37: error: template argument 1 is invalid
 typedef std::weak_ptr<ModelInterface> ModelInterfaceWeakPtr;
                                     ^
/home/ros/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:80:60: error: invalid type in declaration before ‘;’ token
 typedef std::weak_ptr<ModelInterface> ModelInterfaceWeakPtr;
                                                            ^
CMakeFiles/urdf.dir/build.make:62: recipe for target 'CMakeFiles/urdf.dir/src/model.cpp.o' failed
make[2]: *** [CMakeFiles/urdf.dir/src/model.cpp.o] Error 1
CMakeFiles/Makefile2:516: recipe for target 'CMakeFiles/urdf.dir/all' failed
make[1]: *** [CMakeFiles/urdf.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
`

clalancette, 2017-03-13:
This seems to compile fine for me on Debian Jessie.  Can you give a few more details of your setup:

1.  What branch of robot_model are you using?
2.  Are you on an x86_64 machine, or are you using something like Raspbian?
3.  What exact command-line are you using to build?
4.  What happens if you back out the C++11 changes?

ita93, 2017-03-14:
Thanks for you reply.
1. The branch that I used is indigo_devel.
2. I am using an x86_64 machine.
3. I am building catkin Workspace with following command:
$sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/indigo
Sorry, I found that I use source get from command: 
```
$ rosinstall_generator desktop --rosdistro indigo --deps --wet-only --exclude roslisp --tar > indigo-desktop-wet.rosinstall
$ wstool init src indigo-desktop-wet.rosinstall
```
I found that the robot_model source in src folder is differ from source in github, so I checkout the newest commit on indigo-devel branch and copy it into src folder, I also add c++11 flag to CMakeList.txt file, it change to another error:
`[ 50%] [100%] Building CXX object CMakeFiles/urdf.dir/src/model.cpp.o
Building CXX object CMakeFiles/urdf.dir/src/rosconsole_bridge.cpp.o
In file included from /usr/local/include/urdf_parser/urdf_parser.h:45:0,
                 from /home/phind/ros_catkin_ws/src/robot_model/urdf/src/model.cpp:43:
/usr/local/include/urdf_world/types.h:48:41: error: conflicting declaration ‘typedef class std::shared_ptr<urdf::ModelInterface> urdf::ModelInterfaceSharedPtr’
 typedef std::shared_ptr<ModelInterface> ModelInterfaceSharedPtr;
                                         ^
In file included from /home/phind/ros_catkin_ws/src/robot_model/urdf/include/urdf/model.h:43:0,
                 from /home/phind/ros_catkin_ws/src/robot_model/urdf/src/model.cpp:37:
/home/phind/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:76:28: note: previous declaration as ‘typedef class boost::shared_ptr<urdf::ModelInterface> urdf::ModelInterfaceSharedPtr’
 URDF_TYPEDEF_CLASS_POINTER(ModelInterface);
                            ^
/home/phind/ros_catkin_ws/devel_isolated/urdf/include/urdf/urdfdom_compatibility.h:52:34: note: in definition of macro ‘URDF_TYPEDEF_CLASS_POINTER’
 typedef boost::shared_ptr<Class> Class##SharedPtr; \
                                  ^
CMakeFiles/urdf.dir/build.make:54: recipe for target 'CMakeFiles/urdf.dir/src/model.cpp.o' failed
make[2]: *** [CMakeFiles/urdf.dir/src/model.cpp.o] Error 1
CMakeFiles/Makefile2:776: recipe for target 'CMakeFiles/urdf.dir/all' failed
make[1]: *** [CMakeFiles/urdf.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'urdf': 
  Command '['/opt/ros/indigo/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/phind/ros_catkin_ws/build_isolated/urdf && /opt/ros/indigo/env.sh make -j4 -l4

`
Did the step that I do is valid or not?
Could you please help me solve it, thanks you very much

EscontrelaAlejandro, 2017-03-29:
Were you able to find a solution for this problem? Im encountering this error also.

clalancette, 2017-03-31:
Let me see what we can do here.

@ita93 It looks like you are using an old, otherwise out-of-date version of liburdfdom-dev.  That accounts for the problems you reported on March 14.  For that matter, that may be the root cause of all of your problems.  How did you install liburdfdom-dev?  Did you build from source?  If so, what version did you build?

Just FYI, you can get binary packages for ROS indigo on jessie by adding:

```
deb http://packages.ros.org/ros/ubuntu jessie main
```

To /etc/apt/sources.list.  It might help to remove your custom compiled liburdfdom-dev, and install the binary package.

languitar, 2017-04-06:
I am also facing this issue on archlinux. urdfdom is version 1.0.0. This is the latest release that I can find. Is a master version required?

languitar, 2017-04-06:
Could it be that an `#include <urdf_worl/types.h>` is missing in the code?

clalancette, 2017-04-07:
@languitar No, urdfdom 1.0.0 should work fine, I think.  It's possible that there is something missing in the code, but I haven't been able to reproduce it at all, so I'm not sure.  Can you describe your OS setup, environment, and catkin workspace setup?  Also, what's the exact error you are getting?  If I can get all of that, I can attempt to reproduce the problem.

ita93, 2017-04-07:
@clalancette Hi, sorry for my lately reply. I uninstall everything relate to the ROS on my Debian, then I re-download and install again, and it works. When I install ROS before, I follow the guide for Debian Wheezy (guide for jessie is not available), and it have a step to resolve Dependencies not available in the Debian Wheezy branch. When I re-install ROS, i found that these packages are available on Jessie, so I ignore these steps, and It works. The install procedure currently is same as the guide for install ROS on Raspbian Jessie (but not use git reset --hard 9aed725 command)

languitar, 2017-04-09:
It's archlinux 64 and I have been using the [`ros-indigo-*` packages from AUR](https://aur.archlinux.org/packages/?O=0&SeB=nd&K=ros-indigo&outdated=&SB=n&SO=a&PP=50&do_Search=Go) for installing everything.

wallEVA96, 2017-04-15:
hi @ita93  its just do it like the guide on ros.org? then not use git reset -- hard 9ae725 command?

wallEVA96, 2017-04-15:
@ita93 I have do like what u SAY  not use git reset but still not work    so sad   can U help me check?

sloretz, 2017-04-17:
possibly related to 
ros/robot_state_publisher#73
ros/urdfdom_headers#32

clalancette, 2017-08-21:
I think the original issue here has been solved, so I'm going to close this out.  Feel free to reopen if you disagree.

closed 2017-08-21 by clalancette