ros/robot_model #190: joint_state_publisher fails to compile with missing depend on rostest

tspicer01, 2017-03-20:
Hello,

Your recent changes to add the rostest dependeny is causing some compilation errors in our system.  It seems that in joint_state_publisher you missed adding rostest as a dependency in package.xml

Error that we are seeing using the latest on kinetic-devel

Errors << joint_state_publisher:cmake /tmp/buildd/workspace/debian/tmp/build/logs-sdk/joint_state_publisher/build.cmake.000.log
 
  Could not find a package configuration file provided by "rostest" with any
  of the following names:
 
     rostestConfig.cmake
     rostest-config.cmake
 
   Add the installation prefix of "rostest" to CMAKE_PREFIX_PATH or set
  "rostest_DIR" to a directory containing one of the above files.  If
   "rostest" provides a separate development package or SDK, be sure it has
   been installed.

Could you please add rostest as a dependency?

Thank you

clalancette, 2017-03-21:
Just for my own knowledge, what's the configuration that you are using that causes the build failure?  In my own tests here, if I compile the current code with ros-kinetic-rostest installed, everything seems to work.  If I remove ros-kinetic-rostest, then of course I get the failure, but that is expected.  I'm trying to figure out how to replicate your failure.  Thanks!

clalancette, 2017-08-21:
No response here for a while.  I'm going to close this out, feel free to reopen if you need further help.

closed 2017-08-21 by clalancette