ros/robot_model #195: Split this repo and deprecate robot_model

sloretz, 2017-04-24:
Summary of meeting:

**Original issue**: collada_urdf depends on geometric_shapes which depends on octomap. During the ROS Lunar release packages depending on packages in robot_model were blocked needlessly by the release of octomap and geometric shapes.

**Plan**

* Move collada_parser and collada_urdf together to a new repo
* Move kdl_parser and kdl_parser_py together to a new repo
* Move joint_state_publisher to a new repo
* Move urdf and urdf_parser_plugin together to a new repo
* Add a README for this repo which points to the new repositories
* Leave the metapackage robot_model in this repo, but re-release to lunar with status end-of-life
* Update ROS Lunar migration notes on ROS wiki
* Add a new rep like REP-142 that lists each package independently in the robot_metapackage instead of robot_model
* Don't release robot_model metapackage into ROS M

sloretz, 2017-04-24:
@clalancette @mikaelarguedas @wjwwood Does this look right?

sloretz, 2017-04-24:
Closing in favor of https://osrfoundation.atlassian.net/browse/ROS-38

wjwwood, 2017-04-24:
I think it can be split into categories based on what needs to happen by when:

- Before Lunar Release (ideally before Beta):
  - Leave the metapackage `robot_model` in this repo, but re-release to lunar with status end-of-life
    - Make sure the EOL message indicates that the `robot_model` metapackage is deprecated, the other packages have just been moved.
  - Update ROS Lunar migration notes on ROS wiki
    - Mention that the robot_model metapackge has been deprecated
- Any time, but sooner than later:
  - Move `collada_parser` and `collada_urdf` together to a new repo
  - Move kdl_parser and kdl_parser_py together to a new repo
  - Move joint_state_publisher to a new repo
  - Move urdf and urdf_parser_plugin together to a new repo
  - Add a README for this repo which points to the new repositories
- Any time before M-Turtle
  - Add a new rep like REP-142 that lists each package independently in the robot_metapackage instead of robot_model
  - Don't release robot_model metapackage into ROS M

Also, I think might be useful for these to be on GitHub since Jira is not public.

wjwwood, 2017-04-24:
@sloretz it should be fine to have the overarching issue in JIRA as long as the issues are in GitHub, though I do still think having an organizing issue in GitHub helps with cross-referencing for external observers.

wjwwood, 2017-04-24:
Edit to the previous post:

I was just clarifying why I asked for it to be left open here, you don't need to change anything else.

sloretz, 2017-04-24:
@wjwwood The first post is now split according to the categories you listed.

wjwwood, 2017-04-24:
👍 lgtm

mikaelarguedas, 2017-04-25:
@bmagyar FYI, thanks for reporting the issue

sloretz, 2017-04-26:
Step 1 in ros/rosdistro#14675

sloretz, 2017-04-26:
Migration notes updated for lunar http://wiki.ros.org/lunar/Migration

sloretz, 2017-04-28:
Ideas for new repo names?

* **ros/collada_model**
    * collada_parser
    * collada_urdf
* **ros/kdl_model**
    * kdl_parser
    * kdl_parser py
* **ros/joint_state_publisher**
    * joint_state_publisher
* **ros/urdf_model**
    * urdf
    * urdf_parser_plugin

@mikaelarguedas  if these names look ok or if you've got better ones would you mind forking robot_model into them? I don't have the permissions needed to create a repo in the ros org.

wjwwood, 2017-04-28:
My initial reaction is that the `_model` isn't always necessary, though I don't feel strongly about it. I would have done:

- ros/collada_urdf
  - because collada_urdf is the important package, and collada_parser is a means to an end imo
- ros/kdl_parser
  - this seems to fit better (to me at least) than `_model` which doesn't seem to have anything to do with the package names within the repository
- urdf
  - urdf_parser_plugin is part of urdf in my mind, so urdf seems appropriate
  - if I consider what `_model` means when attached to `urdf` I can't come up with a meaning that tells me more than just `urdf`

These are all just suggestions. I think whatever you do will be just fine.

mikaelarguedas, 2017-05-01:
@sloretz Sorry for the late reply. I +1'ed @wjwwood's suggestion but forgot to change your rights. You should know have the rights to create new repos on the ros org unit.

sloretz, 2017-05-04:
`collada_urdf` and `collada_parser` are now at https://github.com/ros/collada_urdf

Self note: Here were the steps to move a package from one repo to another

1. Create the new repo (ex: https://github.com/ros/collada_urdf)
    1. Using the github GUI create an empty repo
    2. Checkout the current repo
    3. add a git-remote for the new repo
    4. push the desired branches to the new repo (ex: indigo-devel, kinetic-devel)
2. Create a new release repo (ex: https://github.com/ros-gbp/collada_urdf-release)
    * initialize it with an empty README
3. Release the packages from the new repo
    1. catkin_generate_changelogs won't work. Add the `Forthcoming` bit to the changelogs manually
    2. catkin_prepare_release works as usual
    3. bloom-release works up to the point of opening the rosdistro PR
        * bloom-release will prep the release repository then close after an assertion about a duplicate package in the distrobution.yaml fails
        * Manually create a PR to update the distrobution.yaml for each rosdistro. The PR should add the new repo and remove the moved packages from the old repo.
4. Delete the packages from the source repo

mikaelarguedas, 2017-05-04:
Example of what the resulting rosdistro PR looks like: https://github.com/ros/rosdistro/pull/14899

John-Nagle, 2017-06-06:
If all this was done, why am I getting this install conflict with ROS Lunar?

https://github.com/ros/robot_model/issues/203

davetcoleman, 2017-06-20:
The state of URDF code locations was already complicated before this split - could you please document these changes on the URDF wiki similar to how I attempted to do so in 2013 [here](http://wiki.ros.org/urdf)? Its amazing to me how many robotics projects (non-ROS, private) around the world use URDF, yet how messy it is to use.

The source for the diagram I made is [here](https://github.com/ros/robot_model/tree/kinetic-devel/robot_model/documentation)

Thanks!!

Alessandro-Barbieri, 2017-06-29:
hello (?)
some packages still pull in `robot_model/collada_parser`
I deleted the `*/collada_parser` folders in `src`, `devel_isolated` and `build_isolated`
run `rosdep update`, regenerated the `.rosinstall` file, run `wstool merge` but `wstool update` continue to redownload `robot_model/collada_parser` and then causing conflicts at build time

how can I fix?

clalancette, 2018-03-19:
All of the work for this is now merged (mostly thanks to @sloretz , thanks).  I'm going to close this ticket out now.

closed 2018-03-19 by clalancette

mikaelarguedas, 2018-03-19:
@clalancette should follow-up issues be opened for the comments that came up on this issue during refactoring ? (e.g. https://github.com/ros/robot_model/issues/195#issuecomment-309756455)

clalancette, 2018-03-19:
> @clalancette should follow-up issues be opened for the comments that came up on this issue during refactoring ? (e.g. #195 (comment))

Good idea, I've opened up https://github.com/ros/robot_model/issues/224 to track that issue.