ros/robot_model #203: ROS Lunar version conflict with Ubuntu 16.04LTS - libcollida-dom2

John-Nagle, 2017-05-31:
ROS Lunar has only one install conflict with Ubuntu 16.04LTS, involving libcollida-dom2. The system on which the install was done was running Ubuntu 16.04LTS, updated with the latest Ubuntu updates before ROS install was attempted. Previous versions of ROS had been removed.

During ROS install, Ubuntu reported a system problem:

```
Package: libcollida-dom2.4-do0 2.4.4+ds1-1.
collida-dom

"An Ubuntu package has a file conflict with a package that is not a genuine Ubuntu package".
```

At the end of the ROS install, that install reported:
```
Errors were encountered while processing:
 /var/cache/apt/archives/libcollada-dom2.4-dp0_2.4.4+ds1-1_amd64.deb
 /var/cache/apt/archives/libcollada-dom2.4-dp-dev_2.4.4+ds1-1_amd64.deb
```

This looks a lot like Issue 118 from 2015: https://github.com/ros/robot_model/issues/118

See also  http://answers.ros.org/question/222892/collada-dom24-sp-dev-causes-dist-upgrade-to-fail-on-ubuntu-1404/

That bug was in ROS-Indigo for Ubuntu 14.04. It was fixed there. Looks like it's been broken again.

John-Nagle, 2017-06-01:
[image]
This broke Ubuntu's package system.

John-Nagle, 2017-06-06:
Here's the problem. There's a conflict with OpenHRP, another robot simulator.

apt-cache showpkg collada-dom2.4-dp-base

> Package: collada-dom2.4-dp-base
> Versions: 
> 2.4.4.7-ubuntu1~trusty1 (/var/lib/dpkg/status)
>  Description Language: 
>                  File: /var/lib/dpkg/status
>                   MD5: 1533959e200afc272e231e3d976fff2e
> 
> 
> Reverse Depends: 
>   openhrp,collada-dom2.4-dp-base 2.4.0.16-ubuntu1~precise1
> Dependencies: 
> 2.4.4.7-ubuntu1~trusty1 - libboost-filesystem1.54.0 (0 (null)) libboost-system1.54.0 (0 (null)) libc6 (2 2.14) libgcc1 (2 1:4.1.1) libstdc++6 (2 4.6) liburiparser1 (2 0.6.0) libxml2 (2 2.7.4) zlib1g (2 1:1.1.4) collada-dom2.4-sp-base (0 (null)) 
> Provides: 
> 2.4.4.7-ubuntu1~trusty1 - 
> Reverse Provides: 
> N:

sloretz, 2017-06-14:
@John-Nagle Where did this output come from? It looks like it misspells coll**A**da.

```
Package: libcollida-dom2.4-do0 2.4.4+ds1-1.
collida-dom
```

sloretz, 2017-06-27:
Hi @John-Nagle, I think rdiankov/collada-dom#26 is the best place, so I'll close this issue. Feel free to reopen.

closed 2017-06-27 by sloretz